negan07/ancistrus #80: Internet LED state with monolithic img #724

Linden1, 2021-02-15:
I was glad to see this monolithic released, I've been using the script installed over V1.0.1.78_1.0.1 for a while and wanted for the the extra rootfs space.
I know it's just a little niggle but my Internet LED remains off with a good net connection. I've proved the LED and logic using `/usr/sbin/led_app on 20` with no problem. I've even gone back to the normal V1.0.1.78_1.0.1 and it came back.
Is this just a mapping error in the code?

Linden1, 2021-02-15:
Just need to add. Once the led is on using led_app it blinks as normal with Internet activity.

negan07, 2021-02-15:
> I've been using the script installed over V1.0.1.78_1.0.1
which command ?

Linden1, 2021-02-16:
> > I've been using the script installed over V1.0.1.78_1.0.1
> 
> which command ?

I used the script `startup.sh`
But that's not the issue. When I flash the new  monolithic image the Internet LED dies. Both orange and white state won't work until I wake the LED manually with `led_app` over telnet.

Linden1, 2021-02-16:
Sorted it.

1. Hard reset
2. Flash D7000-V1.0.1.74_1.0.1.img
3. Hard reset
4. Flash D7000-V1.0.1.78_1.0.1.img
5. Hard reset
6. Flash D7000-V1.0.1.78_1.0.1-ancistrus_negan07-build-724.img
7. Hard reset
8. Setup as normal

Bingo. I thought something my side had gone wrong with the flash with the lack of replies.
All good.

closed 2021-02-16 by negan07