negan07/ancistrus #81: Model check fail due to unexpected string

Extreme79, 2021-05-08:
Hi.
Searching around for unavailable alternative firmware, I found your optimization for the stock that I want to try.
I'm on fw V1.0.1.78_1.0.1 and running sturtup.sh I got the warning for a difeerent hw version.
Manually checking the string returned by _system get dev_name 2>/dev/null_ I see "NightHawk" instead the expected D7000.
I pretty sure it's a V1 (no info on the back, but it's prety old and the latest fw for v2 hw is 1.0.0.66).
I don't know if this project is abandoned or not.. but I'll go with this issue in case of your interest :-)

Linden1, 2021-05-08:
I think you may have changed the router name to 'NightHawk' in the settings? Change it back to 'D7000' and try again.

Extreme79, 2021-05-08:
Hi
I don't remember, maybe possibile, later I'll check.
Anyway it's not a real problem. I solved for me in the geekest way:
I edited the script file directly with vi in telnet shell and executed it 😬
👍
---
Simone Biagini

closed 2021-05-15 by negan07