negan07/ancistrus #82: Latest firmware update failed

Neonik75, 2021-06-14:
before start, thank you very much for your great work.

I have the following data:
Build Revision Author Number: | 724
Building Date: | 2021-02-14 11:57 UTC

the router is set as ACCESS POINT

i tried update firmware procedure 3 times via administration selection file and select you latest .img file
the procedure start and finish correctly but after restart the build revision is always 724 and not 827

the problem could be the router set as ACCESS POINT?
i have to make a reset before firmware update?

thx in advanced for your help

sandrous, 2021-06-15:
“

  *   In case of Ancistrus Home gui webpage old version still cached by the browser, click the ancistrus/Core Setup/Utilities/Refresh GUI button, or refresh the internal htm page frame. (Note that some Safari browsers may require a complete cache page cleanup in addition).
“
Inviato da Posta<https://go.microsoft.com/fwlink/?LinkId=550986> per Windows 10

Neonik75, 2021-06-18:
resolved. thx

closed 2021-06-18 by Neonik75