negan07/ancistrus #83: Updated Netgear firmware: V1.0.1.80

ComputaBloke, 2021-08-23:
Current Version | V1.0.1.78
New Firmware Version | V1.0.1.80

Release Notes
[Security Fixes] Fixed security issues. For more information about security vulnerabilities, visit https://www.netgear.com/about/security.

ComputaBloke, 2021-08-23:
Hmm... seems latest on Product support site is still Firmware Version 1.0.1.78...
https://www.netgear.com/support/product/D7000
https://kb.netgear.com/000061832/D7000-Firmware-Version-1-0-1-78
https://www.downloads.netgear.com/files/GDC/D7000/D7000_FW_V1.0.1.78_1.0.1.zip

So unless there is another channel to access this material through, I guess put this on hold until it's made available by Netgear?

tux-penguin, 2021-08-31:
I installed accidentally V1.0.1.80_1.0.1 and now I lost my access to debug mode and all the ancistrus benefits... Is there a way to downgrade it?

Linden1, 2021-09-01:
> I installed accidentally V1.0.1.80_1.0.1 and now I lost my access to debug mode and all the ancistrus benefits... Is there a way to downgrade it?

Just flash D7000-V1.0.1.78_1.0.1-ancistrus_negan07-build-827.img over the top. Select yes when you get the earlier firmware warning. Don't forget to factory reset.

tux-penguin, 2021-09-01:
> Just flash D7000-V1.0.1.78_1.0.1-ancistrus_negan07-build-827.img over the top. Select yes when you get the earlier firmware warning. Don't forget to factory reset.

Thanks Linden1! That worked. Do I need to factory reset as I can see everything works as it was previously. I had ancistrus up to date before upgrading V1.0.1.80_1.0.1.

Linden1, 2021-09-01:
> Thanks Linden1! That worked. Do I need to factory reset as I can see everything works as it was previously. I had ancistrus up to date before upgrading V1.0.1.80_1.0.1.

Personal choice but standard procedure for me with such a firmware change. Or am I just paranoid :D

negan07, 2021-09-23:
As happened with 1.0.1.70, firmware GPL source code seems to be unreachable due to wrong label name stored on the servers.

Sent a report on [netgear board](https://community.netgear.com/t5/DSL-Modems-Routers/D7000-firmware-1-0-1-80-GPL-Source-Code-Missing/m-p/2143570#M31603)

If some alternative good link is available, please also report.

Linden1, 2021-09-23:
https://www.downloads.netgear.com/files/GPL/R065-A-2018-0247_D7000v1%20GPL.tgz should do it

negan07, 2021-09-24:
> I installed accidentally V1.0.1.80_1.0.1 and now I lost my access to debug mode

It seems that setup.cgi debug mode capability todo mode has been code-removed so utelnetd can be started directly from gui any more.

The alternative method could be `telnetenabled` as explained [here](https://openwrt.org/toh/netgear/telnet.console)

f54rog, 2021-09-25:
> As happened with 1.0.1.70, firmware GPL source code seems to be unreachable due to wrong label name stored on the servers.
> 
> Sent a report on [netgear board](https://community.netgear.com/t5/DSL-Modems-Routers/D7000-firmware-1-0-1-80-GPL-Source-Code-Missing/m-p/2143570#M31603) .
> 
> If some alternative good links are available, please also report.

Hay Negan   
Download with the following  link using Opera

https://www.downloads.netgear.com/files/GPL/D7000_V1.0.1.80_1.0.1_gpl.tgz

negan07, 2021-09-25:
these are the differences between the new and old fw:

```
--- D7000_V1.0.1.78_WW_src/Source/apps/mini_httpd-1.17beta1/mini_httpd.c
+++ D7000_V1.0.1.80_WW_src/Source/apps/mini_httpd-1.17beta1/mini_httpd.c
@@ -2132,8 +2132,9 @@
     }
 
 #ifdef PNPX
-    if (strcasecmp(method_str, get_method_str(METHOD_GET)) == 0 && 
-        strstr(path, "todo=PNPX_GetShareFolderList"))
+    if((strncmp(path, "/setup.cgi?todo=PNPX_GetShareFolderList",strlen("/setup.cgi?todo=PNPX_GetShareFolderList"))==0) 
+	&& (strcasecmp(method_str, get_method_str(METHOD_GET)) == 0) 
+	&& !strstr(path,"htm"))
     {
         need_auth = 0;
     }
```

referring to this article:
https://ssd-disclosure.com/tag/remote_code_execution/

The reason of excluding debug mode & telnet access from setup.cgi code has been made for security prevention in advance maybe but it wasn't strictly mandatory after fixing above..

negan07, 2021-09-29:
Since V1.0.1.80 firmware version, debug mode console access activation from web gui and telnetenabled udp magic packet, have been removed for preventive security reasons ( [read](https://ssd-disclosure.com/tag/remote_code_execution/) ), avoiding the possibility of initializing the router for package installations.

[rescue .img firmware](https://negan07.github.io/ancistrus/tools/ancistrus-arm-D7000/D7000-V1.0.1.80_1.0.1_rescue.zip)

This is the vendor .img firmware, recompiled with kernel & flag optimizations and garbage file removals, including: 
`/usr/sbin/rc_app/rc_debug_mode
`startup service script, and clearing some web gui connection recap webpages, originally showing router's http password unencrypted.

Telnet daemon is activated as default setting.
To prevent eventually further similar security leaks, the telnet daemon can be started only if https router web gui remote login is disabled ( `webGUI-->Advanced-->Advanced Setup-->Remote Management` ).

To deactivate it permanently (factory reset required to reactivate it), type:

```
nvram set telnet_enable=0
rc debug_mode stop
```
To deactivate it temporarly (until next reboot), type:
`rc debug_mode stop`

No remote telnet access is possible because the connection would be not encrypted and unsafe.

negan07, 2021-09-29:
The package `mini_httpd` have also been updated to `1.30-133`, including the fixes above.
Updating this package, firmwares 1.0.1.78 and 1.0.1.80 can be considered equal.

closed 2021-10-13 by negan07