negan07/ancistrus #88: PPPD Package Update Request

Discovery00, 2022-03-03:
Hi, i have got a problem with my service provider. They changed Ericson SSG  to ZTE SSG. I can't connect to the internet. ZTE SSG only supports MS-CHAP V2 and CHAP authentication. Negan, Does the D7000 support them? Please Upgrade PPP daemon in this firmware (pppd 2.4.9 is the latest one). D7000 is legendary your firmware works perfectly 👍🏻 Waiting for your support.. Thx

negan07, 2022-03-08:
As looked, chap v2 support should be already included: are you sure this to be related to chap auth failure ? CAn you give some more infos related like syslog event msgs other than dmesg or any other attempts with similar different router models ?

a connection reconfig would be also recommended if isp has changed auth

Discovery00, 2022-03-08:
Yes i tested with 3 Modems. They connected succesfully. Asus DSL-AC51U has PPP-2.4.7.

Discovery00, 2022-03-08:
Negan, When I convert D7000 to Bridge mode, Asus connected succesfully with Ethernet WAN Connection.

Discovery00, 2022-03-08:
I looked at the Tplink VR2800 GPL source codes. It has the same kernel version as the D7000. And pppd version is 2.4.5. They also added the rp-ppoe package to the PPPD package. No external rp-ppoe plugin. I am attaching the file, you can review it
[ppp-2.4.5.zip](https://github.com/negan07/ancistrus/files/8207050/ppp-2.4.5.zip)
.

negan07, 2022-03-08:
[ppp.zip](https://github.com/negan07/ancistrus/files/8209120/ppp.zip)

these are the differences between 2.4.1 & the one included in the router source code to mantain broadcom/sercomm compatibility

they should be applied to the newer versions
shadow support & plugins other than pppoe & pppoa can be omitted

negan07, 2022-03-08:
> I looked at the Tplink VR2800 GPL source codes. It has the same kernel version as the D7000. And pppd version is 2.4.5. They also added the rp-ppoe package to the PPPD package. No external rp-ppoe [...]

Unfortunately other vendor's code cannot be used easily because code is always heavily customized for the own device
netgear code seems to have 3 level of development in order:

> broadcom
> sercomm
> netgear own

any of them can be different

rp-pppoe is present as another package and could also be updated like atm libs

Discovery00, 2022-03-11:
Negan i tried to compile it (TP-Link ppp2.4.5)  But failed. Maybe can be succes with your old diffs (apps/ppp2.4.1.ppoe4.orig). I will try it. PPP official website have clean source code. Can you compile for Netgear?  Or can you create an ipk file? We can install it with Ancistrus opkg manager. Thanks.

https://download.samba.org/pub/ppp/

Discovery00, 2022-03-11:
Look at your ppp.zip. Search "blacklist" keyword in diffs. Netgear/sercomm packages sends different byte (long) athentication request. This package my nightmare...

Discovery00, 2022-03-11:
you posted diffs 2.4.9 while i was dealing with 2.4.5 wow 👍🏻

Discovery00, 2022-03-11:
I can work with 2.4.9. I will provide feedback

Discovery00, 2022-03-11:
[image]
[image]

Discovery00, 2022-03-11:
Please look at this... This is my problem.
[image]

Discovery00, 2022-03-12:
[image]
PPP 2.4.9 doesn't work.

Discovery00, 2022-03-12:
[ppp-2.4.1-Edited.zip](https://github.com/negan07/ancistrus/files/8237990/ppp-2.4.1-Edited.zip)
Negan Please Compile This :)) Edited 2.4.1

negan07, 2022-03-12:
in the previous I forgot to enable plugin flag on pppd Makefile my mistake

[ppp-249-ng.zip](https://github.com/negan07/ancistrus/files/8238590/ppp-249-ng.zip)
Did some tunings adding libpppoe.so too

pppoa & pppoe plugins remain equal to vendor package

`pppd` has some flags turned off not present on vendor package `tdb shadow mppe multilink` and also serial console interface disabled too (don't think useful now)

to give a try, stop the connection with:
```
rc wan down
rc wan stop

```
backup & replace original files on related locations
`/usr/sbin/pppd` `/lib/libpppoe.so` `/lib/libpppoatm.so`

then restart connection with:

`rc wan restart 1` or either with web gui on the related webpages

a ppp log should be available on `/tmp/ppp_log` location

negan07, 2022-03-12:
> [ppp-2.4.1-Edited.zip](https://github.com/negan07/ancistrus/files/8237990/ppp-2.4.1-Edited.zip) Negan Please Compile This :)) Edited 2.4.1

this is a reverse patch can't be compiled as it is

If wanna avoid MULTI_OFFER flag patch:
[pppd249-nomultiof.zip](https://github.com/negan07/ancistrus/files/8238663/pppd249-nomultiof.zip)

vendor:
[pppd241-nomultiof.zip](https://github.com/negan07/ancistrus/files/8238664/pppd241-nomultiof.zip)

Discovery00, 2022-03-12:
"multi offer" what's does it mean in PPP Rules?

Discovery00, 2022-03-12:
Why does netgear stay so far away from the original ppp source code? What might they be aiming at by placing these restrictions? I sent an e-mail to the Netgear software department. There must be a logical explanation. Thank you too my friend. I've been focusing on this issue for a week. I hope we can come up with a solution.

Discovery00, 2022-03-12:
The plugin here caught my attention. It is very similar to the original codes. But I couldn't adapt the makefiles. I will send you the latest file I made. I also added multi link support.

https://github.com/vicgarin/Actiontec-V1000H/tree/master/bcm963xx_V1000H-31-121L-11/userspace/public/apps/ppp

Discovery00, 2022-03-13:
[ppp-2.4.1.pppoe4.zip](https://github.com/negan07/ancistrus/files/8238727/ppp-2.4.1.pppoe4.zip)
Deleted Multioffer-Inserted Multilink but Makefile Error

Discovery00, 2022-03-13:
I tried the files you sent, they didn't work. I found the source of the problem. The problem is related to the rp-pppoe package. If rp-pppoe is updated, the problem is solved. Netgear and TP-Link rp-pppe/pppoe.h differentiations..
[image]
[image]
[image]

Discovery00, 2022-03-13:
[image]
[image]

negan07, 2022-03-13:
> Why does netgear stay so far away from the original ppp source code? What might they be aiming at by placing these restrictions? I sent an e-mail to the Netgear software department. There must be a [...]

there may be many reasons: security reliability isp specifications & adaptations
Opening a vendor ticket is surely a good option even if it can become slow because they have some support technical levels.

Anyway I think the work made updating 2.4.9 shouldn't be thrown.
Can u confirm problem to be related to pppoe/ptm connection and not chap support?
Can you attach the /tmp/ppp_log diffs from vendor pppd-2.4.1-pppoe.orig and 2.4.9 if any to see if patches have been applied correctly ?

[rp-pppoe_diff.zip](https://github.com/negan07/ancistrus/files/8240254/rp-pppoe_diff.zip)

These are the rpppoe diffs (vanilla version is assumed to be 3.5 even if it should be 3.05 as the latest is 3.15)
As you can see differences are few, mostly logging & casting that can be skipped

Discovery00, 2022-03-13:
Tmp folder not contain PPP-log. I opened PPP debug but same error, 404 file not found.

Discovery00, 2022-03-13:
My ISP says; your DSL modem sending different byte of pppoe request. As you can see in the picture I posted, Netgear is using pppoe v4. tplink uses v8 pppoe. I think the problem is caused by this. Byte lengths are the same. Now it comes to my mind that the internet service provider blocked my dsl modem. Maybe by Mac address or Serial number. Maybe ISP is using non-standard rfc2516 protocol. who knows :) Isp just says, "Change your modem"

Discovery00, 2022-03-13:
Negan pppd changes maybe working. But it didn't work on my line. I would like to try. But isp unfortunately does not give wan ip address. The log file was not created in the modem. But I can say that the original ppp2.4.1 also had a debug log

Discovery00, 2022-03-13:
The modem does not accept this plugin when it makes changes to the files. I saw the md5 value inside the plugin. A conflict may occur when the modification is made.

Discovery00, 2022-03-13:
> Anyway I think the work made updating 2.4.9 shouldn't be thrown. Can u confirm problem to be related to pppoe/ptm connection and not chap support? Can you attach the /tmp/ppp_log diffs from vendor pppd-2.4.1-pppoe.orig and 2.4.9 if any to see if patches have been applied correctly ?
Yes, problem related with Pppoe connection problem. Confirmed.

negan07, 2022-03-14:
rp-pppoe 3.15 with netgear modifications
[pppoe315ng.zip](https://github.com/negan07/ancistrus/files/8245422/pppoe315ng.zip)

rp-pppoe 3.15 original
[pppoe315or.zip](https://github.com/negan07/ancistrus/files/8245424/pppoe315or.zip)

Discovery00, 2022-03-14:
Rp-pppoe 3.15-original and rp-pppoe 3.15 with Netgear modification, both works successfully.

Discovery00, 2022-03-14:
I will send a logs.

Discovery00, 2022-03-14:
[Rp-org_log.txt](https://github.com/negan07/ancistrus/files/8247291/Rp-org_log.txt)
[Rp-ng_log.txt](https://github.com/negan07/ancistrus/files/8247292/Rp-ng_log.txt)

Discovery00, 2022-03-14:
Negan, can you make pppd 2.4.5 without netgear modifications? No multi offer flag and other modifications.

Discovery00, 2022-03-14:
Yes i tried every combination. Rp-pppoe works flawlessly. But modem can't connect to the internet. I have got a good news for you. I found Netgear PPPD 2.4.1 without multi offer flags and etc. Pure plugin in other routers gpl. And I found 2.4.4 in Netgear China firmware gpl :)

Discovery00, 2022-03-14:
Compile these files without modification. Original Netgear plugins.

Discovery00, 2022-03-14:
You can send it as 2.4.1 orig and 2.4.4 orig. I will try . And we will solve this problem. 👍🏻

Discovery00, 2022-03-14:
[ppp-2.4.1-R7000-EU.zip](https://github.com/negan07/ancistrus/files/8248855/ppp-2.4.1-R7000-EU.zip)
[ppp-2.4.4-R7000-CN.zip](https://github.com/negan07/ancistrus/files/8248856/ppp-2.4.4-R7000-CN.zip)
Good Luck

Discovery00, 2022-03-19:
Hi Negan, I found a project for Netgear Dgnd4000.
http://alfie.altervista.org/amod4000/
He succesfully patched pppd 2.4.7 in firmware. Maybe he will help us.

negan07, 2022-03-21:
The files you provided result incompatible because they come from a different source board.
At this point I think that cross-patching from different boards results in a failure approach attempt.

Are you sure that your new isp configuration matches the requirements ?
Have you correctly setup vlan id tag in the connection setup ?
I suggest you to rebuild setting configuration making particular attention to wan settings configuration taking care of ISP specification.
The way of sending an email ticket to netgear support or better on community board support is still valid: if the problem is device-related maybe someone else with the same isp is involved.

negan07, 2022-03-21:
[ppp-2.4.9-orig.zip](https://github.com/negan07/ancistrus/files/8319652/ppp-2.4.9-orig.zip)

this would be pppd-2.4.9 with libpppoe & libpppoatm plugin libs without further vendor modifications

negan07, 2022-04-11:
```
linux-atm
ppp
rp-pppoe
```
packages are on the way: they must be fully tested under various connection modes:
```
atm (adsl)
ptm (vdsl)
```
on their various possible configurations

make sure to mantain a local backup of these files in case of connection failures:
```
libatm.so.1.0.0
atmarp
atmarpd
libpppoatm.so
libpppoe.so
pppd
pppoe
```

Discovery00, 2022-04-15:
Hello,
I tested your Packages. All packages are working without any problems. But my problem still persists :) I'm throwing the log file. The problem is related to Netgear source codes. They are blocking PPP Remote access. Anyway PPPD 2.4.9 ported successfully. Congratulations. Good job 👍 

[PPP_Log.pdf](https://github.com/negan07/ancistrus/files/8498394/PPP_Log.pdf)

Discovery00, 2022-04-15:
Linux-atm 2.5.2 Tested 👍 
PPPD 2.4.9 Tested 👍  
RP-PPPOE 3.15 Tested 👍

negan07, 2022-04-16:
If someone has an atm connection It would be nice to see if all works correctly

Discovery00, 2022-04-16:
I haven't got an ATM connection. Only PTM.

Discovery00, 2022-04-16:
My neighbours also have ATM :)

Linden1, 2022-04-17:
PPPoE is okay over WAN, if that helps.

negan07, 2022-04-17:
so new packages seems to work properly on main connection types

working on your specific failure case is hard because no reproduction is possible to test it

you should try to avoid `rc_wan` method and trying to configure ppp connection manually looking with `ps` the parameter of `ppp` cmd when invocated & trying to setup proper wan id priority

the same could be done with working router making a comparison between both procedures of connection

Discovery00, 2022-04-25:
Hi Negan Are you here :)

negan07, 2022-04-25:
any news about ppp connection ?
New monolithic is about to be pushed because there are many package upgrades: if there is something about this issue could be patched before.

Discovery00, 2022-04-25:
Dude, no progress on ppp. Not connecting. Can I ask you something? Can we update the bridge-utils plugin? :)

Discovery00, 2022-04-25:
Your new software includes some radical changes. Good work. This plugin is pretty old too. Forgotten aside. waiting to die :)

Bridge-utils 1.7.1 Latest one. 
For Source Code:
https://mirrors.edge.kernel.org/pub/linux/utils/net/bridge-utils/

Discovery00, 2022-04-26:
Knock Knock

negan07, 2022-04-27:
Bridge utils has been bumped but not a critical package

Discovery00, 2022-04-27:
That's important for me cuz I want to buy a Openwrt Router bridging with D7000 DSL only mode.

Discovery00, 2022-04-27:
Netgear can't connect on my dsl line u know...

Discovery00, 2022-04-27:
I will test it Negan thx👍🏻👍🏻👍🏻

closed 2022-04-28 by negan07