negan07/ancistrus #90: V1.0.1.84_1.0.1 build #916 will not connect to the Internet

m0lz, 2022-04-30:
Hi Negan07

I have a problem with V1.0.1.84_1.0.1 build #916, it will not connect to the Internet.

If I roll back to V1.0.1.82_1.0.1 build #875 there are no problems.

I have tried flashing the D7000 twice now with the new build, each time also reloading my backup settings, and in both cases the router will not connect to the internet. I flashed build 916, then reflashed 875 and the internet connected. Again flashed 916, no internet again, and then reflashed 875 and the Internet is now working again.

I do not have a corrupt download, and the router settings all seem fine after the flash of 916, it just refuses to connect to the internet and I have an Orange LED for internet connection with 916 installed.

Let me know what you need to look into this 👍

Linden1, 2022-04-30:
I can confirm this issue. PPPoE over ATM was fine when I tested. Now it's
been integrated I can't connect. Wound back to #875 for now.

Linden Gill

negan07, 2022-04-30:
wan connection have been updated

which package is involved ? (more than one possible)

```
ppp
rp-pppoe
brctl
```

Linden1, 2022-04-30:
Going back to issue #88 this is the same ipk I installed on 875 and confirmed was working on PPPoE over WAN(ATM). I'll give it a try on 916

Linden1, 2022-04-30:
Confirming the package in wan_orig_0.0.0_armD7000.zip fixes this issue.
Flash 916, install the package negan07 has linked here and Reboot.

m0lz, 2022-04-30:
> wan connection have been updated on #916
> 
> this package should revert wan to original (unzip & manual install from package manager): installing it there's no need to reflash [wan_orig_0.0.0_armD7000.zip](https://github.com/negan07/ancistrus/files/8596708/wan_orig_0.0.0_armD7000.zip)
> 

Package manager?
Sorry but I only know how to update the firmware with an .img, that .ipk file is in the realms of voodoo magic to me :)

Linden1, 2022-04-30:
[image]

m0lz, 2022-04-30:
Ah I see, so I need to install once again the 1.0.1.84, then use the package manager to overwrite that package and test?

Linden1, 2022-04-30:
> Ah I see, so I need to install once again the 1.0.1.84, then use the package manager to overwrite that package and test?

Yes. It's that simple. Don't forget to Reboot after.

negan07, 2022-04-30:
> Going back to issue #88 this is the same ipk I installed on 875 and confirmed was working on PPPoE over WAN(ATM). I'll give it a try on 916

so the package `ppp-2.4.9` is fully working with atm connection (adsl) ?
any test with ptm ?

please report which test have been made with which connection to isolate the faulty package

Linden1, 2022-04-30:
PPPoE over WAN(ATM) - working 👍

negan07, 2022-04-30:
please report which test have been made with which package and with which connection to isolate the faulty package

e.g. starting from old original wan packages or from #875 build install & reboot once per time `linux-atm, ppp, rp-pppoe, bridge-utils` (even if this last one should be involved...)

m0lz, 2022-04-30:
I just updated the firmware from 875 to 916 again
then installed the wan orig .ipk file
rebooted

And the internet connection is working again
VDSL(PTM) - VDSL(2)
PPPoE

negan07, 2022-04-30:
[ppp_2.4.9-001_armD7000.zip](https://github.com/negan07/ancistrus/files/8596929/ppp_2.4.9-001_armD7000.zip)

If someone wanna try, I've recompiled ppp without any further vendor patch modifications
unzip & manual install

Linden1, 2022-04-30:
Build #875
PPPoE over WAN (ATM) > connecting fine.
install linux-atm 2.5.2 > reboot > connecting fine.
install ppp 2.4.9 > lose connection > reboot > no connection (LED orange).

output from cat /tmp/ppp_log: cat: can't open '/tmp/ppp_log': No such file or directory

Installed wan_orig package > reboot > connecting fine.

Linden1, 2022-04-30:
ppp_2.4.9-001_armD7000 breaks it too :(

Discovery00, 2022-04-30:
I tested 3 packages, they work fine

Discovery00, 2022-04-30:
Please install with package manager

Discovery00, 2022-04-30:
Install Atm first then ppp, rp-pppoe
[image]

Discovery00, 2022-04-30:
https://github.com/negan07/ancistrus/issues/88#issuecomment-1100432170
Looks at this log file :) I installed on build 875 via package manager. Test OK

Linden1, 2022-04-30:
> Install Atm first then ppp, rp-pppoe
> [image]
> 

With 916 all the packages are preinstalled. Overwriting with the wan_orig only changes ppp to an earlier package. So I have them all already, except ppp

Discovery00, 2022-04-30:
I think Bridge-utils causes this problem

Discovery00, 2022-04-30:
Send your ppp log file on build 916 please

Linden1, 2022-04-30:
I've even tried using startup.sh to build from D7000-V1.0.1.84_1.0.1_rescue.img and still this same issue.

Linden1, 2022-04-30:
> Send your ppp log file on build 916 please

still getting the error
can't open '/tmp/ppp_log': No such file or directory

negan07, 2022-04-30:
This is the only package not involved into chap auth connection stage

`brctl` simply creates bridges between interfaces
executing:
`brctl show`
or
`ip link
`or 
`ifconfig`

you should see at least `group1`

The faulty package should be `ppp-2.4.9`

From a clean firmware or from #875 installing all except `ppp-2.4.9` should result in a successful connection

negan07, 2022-04-30:
> > Install Atm first then ppp, rp-pppoe
> > [image]
> 
> With build #916 all the packages are preinstalled. Overwriting with the wan_orig only changes ppp to an earlier package. So I have them all already, except ppp - confirmed with package manager. This issue seems to be different to the one you had in #88

I agree that the issue related refers to a different auth connection problem, also present on original vendor `ppp-2.4.1`

Linden1, 2022-04-30:
> This is the only package not involved into chap auth connection stage
>
> `brctl` simply creates bridges between interfaces executing: `brctl show` or `ip link `or `ifconfig`
>
> you should see at least [...]

It does - confirmed

negan07, 2022-04-30:
It would be useful to investigate if the problem happens with `ptm` (vdsl) connections too
It seems that all the trials here have been done with `atm` (adsl) connection: is it right ?

Linden1, 2022-04-30:
> It would be useful to investigate if the problem happens with `ptm` (vdsl) connections too It seems that all the trials here have been done with `atm` (adsl) connection: is it right ?

for me, yes ATM

negan07, 2022-04-30:
[ppp_2.4.9-002_armD7000.zip](https://github.com/negan07/ancistrus/files/8597807/ppp_2.4.9-002_armD7000.zip)
I have recompiled ppp-2.4.9 with brcm patch support except for the new `libpppoe` remained unchanged (latest one is used)

In case of failure, it would be useful looking for any kind of helpful error messages apart "no connection" e.g. in the router log message events on advanced gui or http://192.168.0.1/ppp_log or /tmp/ppp_log if any

Discovery00, 2022-04-30:
http://192.168.0.1/?todo=ppp_debug

Discovery00, 2022-04-30:
First of all, you must be enable ppp debug.

Discovery00, 2022-04-30:
Greetings :)

Discovery00, 2022-04-30:
Please try enabling this. then send your ppp log file

Discovery00, 2022-04-30:
Negan, Netgear Closed PPP debug function in latest firmware. Do you know? I cant enable on Original 0.84 release.

Linden1, 2022-04-30:
ppp_2.4.9-002_armD7000

Working for ATM on build 619 😬

m0lz, 2022-05-01:
Just come home from night shift, I need to find out if my daughter is working remotely from home today first, if not then I will test ppp_2.4.9-002_armD7000 in a few hours on VDSL

m0lz, 2022-05-01:
VDSL

Installed Build #916
Rebooted
Installed ppp_2.4.9-002_armD7000.zip with package manager
Rebooted

Internet connection does not connect.

There is a behaviour difference, the LED now keeps repeatedly retrying whereas previously it was just solid Orange, now it will blink white 6 times and then go Orange again, and then repeat. However no matter how many times it retries I still do not get any internet connectivity.

Log reads as follows :

[Internet disconnected]                                          
[Internet connected] IP address: (null)                                          
[Internet disconnected]                                          
[Internet connected] IP address: (null)                                          
[Internet disconnected]                                          
[Internet connected] IP address: (null)                                          
[Internet disconnected]                                          
[Internet connected] IP address: (null)                                          
[Internet disconnected]                                          
[Internet connected] IP address: (null)                                          
[Internet disconnected]                                          
[Internet connected] IP address: (null)                                          
[Internet disconnected]                                          
[Internet connected] IP address: (null)                                          
[Internet disconnected]                                          
[DHCP IP: (192.168.0.2)] to MAC address 08:84:9D:E4:69:A9                                          
[Internet connected] IP address: (null)                     
[Internet disconnected]                                          
[Internet connected] IP address: (null)                                          
[Internet disconnected]                                          
[Internet connected] IP address: (null)                                          
[Internet disconnected]                                          
[Internet connected] IP address: (null)                                          
[Internet disconnected]                                          
[Internet connected] IP address: (null)                                          
[Internet disconnected]                                          
[Internet connected] IP address: (null)                                          
[Internet disconnected]                                          
[Internet connected] IP address: (null)                                          
[Internet disconnected]                                          
[Internet connected] IP address: (null)                                          
[Internet disconnected]                                          
[Internet connected] IP address: (null)                                          
[Internet disconnected]                                          
[Internet connected] IP address: (null)                                          
[Internet disconnected]                                          
[Internet connected] IP address: (null)                                          
[Internet disconnected]                                          
[DHCP IP: (192.168.0.2)] to MAC address 08:84:9D:E4:69:A9                                          
[Internet connected] IP address: (null)                                          
[admin login] from source 192.168.0.3                                          
[admin login] from source 192.168.0.3                                          
[DHCP IP: (192.168.0.4)] to MAC address 5A:B3:CA:2E:6F:54                                          
[DSL: Up]                                           
[DHCP IP: (192.168.0.13)] to MAC address 02:0F:B5:5F:D1:02                                          
[DHCP IP: (192.168.0.3)] to MAC address A0:E7:0B:FA:27:BF                                          
[DHCP IP: (192.168.0.2)] to MAC address 08:84:9D:E4:69:A9                                          
[Initialized, firmware version: V1.0.1.84_1.0.1 ] 

So it seems to be connecting and then disconnecting.

I have now re-installed wan_orig_0.0.0_armD7000.zip using Package manager over the above installation, rebooted, and regained internet access.

Linden1, 2022-05-01:
Do you think you can try Installed ppp_2.4.9-001_armD7000.ipk?

m0lz, 2022-05-01:
> Do you think you can try with ppp_2.4.9-001_armD7000.ipk?

Just tried, installed Build #916, reboot, installed ppp_2.4.9-001_armD7000.ipk, reboot = Thats worse than ppp_2.4.9-002_armD7000.ipk, I get a solid Orange internet LED, log ...

[admin login] from source 192.168.0.3                                          
[admin login] from source 192.168.0.3                                          
[DSL: Up]                                           
[DHCP IP: (192.168.0.13)] to MAC address 02:0F:B5:5F:D1:02                                          
[DHCP IP: (192.168.0.3)] to MAC address A0:E7:0B:FA:27:BF                                          
[DHCP IP: (192.168.0.2)] to MAC address 08:84:9D:E4:69:A9                                          
[Initialized, firmware version: V1.0.1.84_1.0.1 ]

.. At least with ppp_2.4.9-002_armD7000.ipk it keeps trying to connect with the internet as previously posted.

Now reverted to Build #916 with Wan_Orig installed for a working internet.

negan07, 2022-05-01:
> ppp_2.4.9-002_armD7000
> 
> Working for WAN (ATM) on build 916 grimacing

I wouldn't bet a coin on this but it's good.

ppp-2.4.9-001 is unchanged apart makefiles adaptations for arm-toolchain of course
result:
ATM failure (no chap auth attempt)
PTM failure (no chap auth attempt)

ppp-2.4.9-002 includes broadcom/sercomm/netgear patch adaptations apart pppoe lib plugin
result:
ATM successful
PTM failure (loop on trying to obtain chap auth)

ptm is vdsl on pppoe so it's should be reasonable to obtain an error on it

The original netgear package is a variation of ppp-2.4.1 including (patching) the full libpppoe plugin, in turn patched with broadcom modifications. I need to extract the modifications and to apply them on the 2.4.9 pppoe plugin code because totally different (all the functions & modules are different)

negan07, 2022-05-01:
Please, it would be useful to know if you have the following variables inside your nvram settings, typing:

```
nvram show | grep ppp_terminate

nvram show | grep pppoe_sid

nvram show | grep pppoe_srv_addr
```

Linden1, 2022-05-01:
> Please, it would be useful to know if you have the following variables inside your nvram settings, typing:
> 
> ```
> nvram show | grep ppp_terminate
> 
> nvram show | grep pppoe_sid
> 
> nvram show | grep pppoe_srv_addr
> ```

nvram show | grep ppp_terminate
ppp_terminate=1

nvram show | grep pppoe_sid
{no output}

nvram show | grep pppoe_srv_addr
{no output}

m0lz, 2022-05-02:
I have the same results as Linden1

negan07, 2022-05-02:
[ppp_2.4.9-003_armD7000.zip](https://github.com/negan07/ancistrus/files/8602873/ppp_2.4.9-003_armD7000.zip)

In this attempt I've include both lib pppoatm & libppoe again with some modifications on makefiles removing new stuffs not used by original vendor code

If not working, a debug on the fly should be done...

those who want to try: take care of `pppd` command line you have on `ps | grep pppd` after trying to connect (it should result as `pppd_1` )

The idea is to kill pppd then re invoking pppd adding the debug flags to see some screen msg if they can help

Linden1, 2022-05-02:
I have spent 2hrs collecting data for you. I've disconnected so many times I can no longer get a connection, even with build 875.
The following results for ppp_log are for a freshly installed (build) 916 then the various ipk installed.

<b>Bare build 916:</b>
  setting line discipline hook
  Sending PADI
  Clear blacklist.
  Clear blacklist.
  Clear blacklist.
  Clear blacklist.
  Clear blacklist.
  setting line discipline hook
  Sending PADI
  Clear blacklist.
  Clear blacklist.
  Clear blacklist.
  Clear blacklist.
  Clear blacklist.
  setting line discipline hook
  Sending PADI
  Clear blacklist.
  Clear blacklist.
  Clear blacklist.
  Clear blacklist.
  Clear blacklist.

<b>ppp_2.4.9_003_armD7000</b>
  setting line discipline hook
  Sending PADI
  Clear blacklist.
  Clear blacklist.
  Clear blacklist.
  Clear blacklist.
  Clear blacklist.
  setting line discipline hook
  Sending PADI
  Clear blacklist.
  Clear blacklist.
  Clear blacklist.
  Clear blacklist.
  Clear blacklist.
  setting line discipline hook
  Sending PADI
  Clear blacklist.
  Clear blacklist.
  Clear blacklist.
  Clear blacklist.
  Clear blacklist.

<b>ppp_2.4.9_002_armD7000</b>
setting line discipline hook
send PPPOE Discovery V1T1 PADI session 0x0 length 12
 dst ff:ff:ff:ff:ff:ff  src 2c:30:33:30:6e:09
 [service-name] [host-uniq  8e 12 00 00]
Recv PPPOE Discovery V1T1 PADO session 0x0 length 49
 dst 2c:30:33:30:6e:09  src 24:21:24:87:32:57
 [service-name] [AC-name LDSCD2-BNG-C1] [host-uniq  8e 12 00 00] [AC-cookie  1b 02 94 39 2b 82 ea d5 cb a0 ea 55 96 83 44 d4]
  Send PPPOE Discovery V1T1 PADR session 0x0 length 32
dst 24:21:24:87:32:57  src 2c:30:33:30:6e:09
 [service-name] [host-uniq  8e 12 00 00] [AC-cookie  1b 02 94 39 2b 82 ea d5 cb a0 ea 55 96 83 44 d4]
Recv PPPOE Discovery V1T1 PADS session 0x73d length 12
 dst 2c:30:33:30:6e:09  src 24:21:24:87:32:57
 [service-name] [host-uniq  8e 12 00 00]
PADS: Service-Name: ''
PPP session is 1853
Connected to 24:21:24:87:32:57 via interface eth0.1
using channel 4
Using interface ppp1
Connect: ppp1 <--> eth0.1
sent [LCP ConfReq id=0x1 <mru 1492> <magic 0xd826d8f4>]
rcvd [LCP ConfReq id=0x5b <mru 1492> <auth chap MD5> <magic 0x43961177>]
sent [LCP ConfAck id=0x5b <mru 1492> <auth chap MD5> <magic 0x43961177>]
rcvd [LCP ConfAck id=0x1 <mru 1492> <magic 0xd826d8f4>]
mtu is <1492> ao->mru is <1492>
Set ppp0 MTU 1492
sent [LCP EchoReq id=0x0 magic=0xd826d8f4]
rcvd [CHAP Challenge id=0x1 <c96bbf5cd4ea1c2164f45436f3df6d9b4081a37512b870ab6c4303ca7a54e9453fa92194>, name = "LDSCD2-BNG-C1"]
sent [CHAP Response id=0x1 <5db14b831f5e12fb12d19f8513da1880>, name = "dsl002295928@broadband.vodafone.co.uk"]
rcvd [LCP EchoRep id=0x0 magic=0x43961177]
rcvd [CHAP Failure id=0x1 "User Object Not Found"]
CHAP authentication failed: User Object Not Found
CHAP authentication failed
sent [LCP TermReq id=0x2 "Failed to authenticate ourselves to peer"]
Script /etc/ppp/ipv6-down started (pid 4766)
rcvd [LCP TermReq id=0x5c]
sent [LCP TermAck id=0x5c]
Script /etc/ppp/ipv6-down finished (pid 4766), status = 0x0
rcvd [LCP TermAck id=0x2]
Connection terminated.

When testing wan_orig it became obvious that my fibre was no longer accepting connections for my D7000, so I won't be much help now until I get that sorted out.
All connections were PPPoE over WAN(ATM)

negan07, 2022-05-02:
too much incompatibility: I think it's better to regress to ppp-2.4.1

 > I've disconnected so many times I can no longer get a connection, even with build 875

This should be an isp security system due to connection failure trials hammering: it should be solved after a while (leasing time maybe)

negan07, 2022-05-02:
updated monolithic #917

m0lz, 2022-05-02:
:) I'll be honest I would have been lost doing the tests Linden1 carried out.

VDSL
Build #917 installed = No problems to report.
Logs seem normal with no errors 👍

negan07, 2022-05-03:
pppd which is the executable file of ppp-2.4.9, is driven by proprietary binaries like `start_ppp`: they create a copy link of themselves called with the suffix name of the vlan id, usually 1 and then run it. This makes hard to debug it

Killing first `start_ppp` then `pppd_1` then reinvoking pppd_1 with parameters adding debug logfile /tmp/ppp_log

The traditional adsl should work with it, the problem is the pppoe lib for vdsl, totally different and hard to follow without the full code

Linden1, 2022-05-03:
> Is Linden1 able to connect again hopefully

Currently connected to my isps router. I'll try setting it up again when I get home.

negan07, 2022-05-03:
try to override the wan mac address setting up another
or the same of the router isp

Lease may require more time, also a mac ban should be temporary

Linden1, 2022-05-03:
I've tried cloning the mac of another (dead) D7000 and the mac from my isp's router, in combination with build #875 and Netgear's official D7000-V1.0.1.84_1.0.1. For now I think this thing is dead, to be resurrected in a couple of weeks.

Linden1, 2022-05-03:
I've tried cloning the mac of another (dead) D7000 and the mac from my isp's router, in combination with build #875 and Netgear's official D7000-V1.0.1.84_1.0.1. For now I think this thing is dead, to be resurrected in a couple of weeks.
One thing with build #917. It won't let me connect by connecting my D7000 WAN to my isp router's LAN using dhcp. It's insisting I need to use a username and password. It's constantly trying to connect with ppp even though it's not needed in that mode.

negan07, 2022-05-03:
Tried yesterday before releasing 917 with ethernet wan mode without issuing. 

`ppp` shouldn't be involved because you are not doing a chap auth in this case until you are using `ipoa` mode
Check the preferences, if the isp allows other vendors routers but it still insist to refuse your connection, ask it for a line reset: sometimes when line conditions drops or many failures happen it may switch you into a failsafe or SOS mode.

Linden1, 2022-05-03:
> Tried yesterday before releasing 917 with ethernet wan mode without issuing.
>
> `ppp` shouldn't be involved because you are not doing a chap auth in this case until you are using `ipoa` mode
> Check [...]

I agree ppp should not be involved, but I was definitely being asked for my username and password which were continually rejected.
So I installed build #916 and added wan_orig and set up a dhcp connection using the wizard. Then I updated to build #917 and the settings have stayed and I'm connected through my D7000 through my isp router. My settings survive reboots.
Contacting my isp is the next step. It's not going to be easy though. Vodafone UK customer service is the worst as any customer will attest.

m0lz, 2022-05-03:
Good luck Linden.
I dont know if this will help but does your ISP have any kind of account settings that typically lock you in to their DNS servers.
I just recall when I got the D7000 to replace the ISP router (objective was so that I could set my own DNS servers for primary and secondary in the router which the ISP router would not allow), I had to go into my ISP account settings and disable a few default setups like BT Parental Control and the ISPs provided anti-malware protection .. And then flush the computer cache and reboot the new router a few times before I could get the internet back. The ISP was unusually quite helpful considering we were discussing a none ISP router which they dont officially support.
Maybe your ISP has defaulted to the new user settings after you were trying to regain access during your testing of the firmware.
I think I am probably teaching you to suck eggs having witnessed the more advanced level of understanding of these things you have already displayed, and this is more than likely well beyond me to try and help with :)

Linden1, 2022-05-03:
You may very well be right.
Yes, my rapid disconnects will likely have upset my isp, and having ruled out my mac being blacklisted I feel it's highly likely they've changed my login. Unfortunately I can't yet read the login info from a vodafone hub so my only recourse now is to speak to them.

negan07, 2022-05-03:
> (objective was so that I could set my own DNS servers for primary and secondary in the router which the ISP router would not allow)

this sometimes happen when voip is involved: the router must use internal dns to match voip aliases not reachable from the outside of the isp network

This is not the situation, but there is a trick usable: mantain isp dns on the router then order the dhcp server to configure some different dns as options when a device ask for connection
e.g.
`option dns 1.1.1.1`

on D7000 this must be done manually modifying rc script (there should be an old issue involved)

Linden1, 2022-05-04:
I'm back online. Vodafone in their infinite wisdom decided to change my login. Why, I don't know. But the recent logs I posted are now redundant.

[11 more replies]

closed 2022-05-09 by negan07