negan07/ancistrus #91: GUI reporting Build 875

Shockeh, 2022-05-03:
Hi ancistrus,

Possible red herring, but when logged in via SSH, all packages show as latest versions, but the Web GUI continues to report Build 875, 2021-11-03 21:22 UTC, and that there's a new Monolithic Firmware Available.

I've --force-reinstalled ancistrus-core just to see if that made any difference, but no luck.

Shockeh, 2022-05-03:
Addendum: I've tried reinstalling mini_httpd without success, after noticing a number of the headings under the Ancistrus panel are now blank:

[image]

m0lz, 2022-05-03:
The "new Monolithic Firmware Available" report may have been due to this issue ...
https://github.com/negan07/ancistrus/issues/90
.. During troubleshooting that issue, there was a build #916 released, which was subsequently removed, and now on conclusion of that issue there is a new build #917
So possibly prior to regression you received the update notice.

Shockeh, 2022-05-03:
Thanks @m0lz - Just to check, I tried reflashing with 917, but the same behaviour recurred, still shows at 875.

Although I did get something new, upon login I got a popup:
`Warning ! Repository update not available !Error code: 1`

negan07, 2022-05-04:
clean the D7000 related cache browser and try again
Repository results up here for me

Which browser do you have in use ?

Linden1, 2022-05-06:
Mine reports build 875 too in Chrome. Look at the date and the firmware version.
[image]
But if I switch to firefox the build and date are correct
[image]
I don't have an issue. Just wanted to support this thread and say I haven't found a way to fix it other than to use firefox. No amount of GUI refreshing or cache clearing fixes it. Just use firefox.

negan07, 2022-05-06:
The file informations are fetch from router firmware's file system, in particular `/usr/etc/svn.info`

If the file is changed it contains updated infos. The browser can't fetch the old file with old infos because they're no more available so it means that it keeps them into its internal cache.

Clearing the cache related to D7000 or 192.168.0.1 will delete also the old page infos.

To clear the cache you can also act manually reloading not the whole pagebut the internal frame only.

Otherwise, use the button `Refresh GUI` present on `ancistrus -> core setup -> utilities`

Linden1, 2022-05-06:
I got it fixed in chrome. I signed out of my Google account, cleared the cache then signed back in and synced again. A strange one but it's definitely not an issue wit 917.

negan07, 2022-05-06:
It's an old browser related one.
I can't force the refresh on the whole index page because it is controlled by vendor cgi and I can't trick it  because no code is available.

I decided to force a refresh in the internal frame with this: 
https://github.com/negan07/ancistrus/commit/6ae89771cd75dcf7f948a93c90acaa9438feee48

`document.getElementById('formframe').contentDocument.location.reload(true);
`
but probably this method doesn't have effects on latest chrome versions, while working on mozilla/gecko, edge & safari

So this report is useful, because I can investigate the reasons on chrome too

negan07, 2022-05-06:
I've just tried with all the typical browsers, altering manually `/etc/svn.info` with another firmware version string or anything else, obtaining a successful page refresh in any case, with router's ip login & just in case also with www.routerlogin.net

At this point I ask:

- the chrome version in use (laptop/mobile also)
- any particular plugin installed related
- the page debug info tool output of the loaded page if you can

Linden1, 2022-05-06:
The error is back.

> * the chrome version in use (laptop/mobile also)
Version 101.0.4951.54 (Official Build) (64-bit) on Windows 10 Laptop
> * any particular plugin installed related
All plugins disabled
> * the page debug info tool output of the loaded page if you can
How do we do this? I'm willing to help.

negan07, 2022-05-06:
the dev tools can be called with `ctrl + shift + I`
`elements` tab
see for errors (some problems & warnings are normally present)

Linden1, 2022-05-06:
[image]
[image]
[image]

Linden1, 2022-05-06:
This is a strange one. Not only is it showing as build 875, it's also showing the update banner for build 917.
[image]

negan07, 2022-05-06:
same for me

If you want to try modifying `/etc/svn.info` on the fly telnet/ssh the router and type:

sed -i "s|917|mumblemumble|" /etc/svn.info

the refresh the page and see

Linden1, 2022-05-06:
sed -i "s|917|mumblemumble|" /etc/svn.info followed by a refresh makes no difference

negan07, 2022-05-06:
how do you refresh ?
clicking ancistrus tab ? or anything else ?

negan07, 2022-05-06:
if you want to repeat the experiment

`sed -i "s|mumblemumble|anything|" /etc/svn.info`

and so on

Linden1, 2022-05-06:
Now it's stuck again. I tried changing it back to 917, cleared cache and cookies and it's still mumblemumble.
I honestly think it's a problem with chrome not clearing the cache properly. I'm going to report this to Google and see what comes of it.

negan07, 2022-05-06:
Do you still obtain 875 even after clearing it all ?
I'm using a portable version up to date of chrome without any gmail account association: can this create stucks ?

Linden1, 2022-05-06:
nevermind. I messed it up. I've managed to change it back to 917 and the refresh works.
But yes, I do think this is actually a chrome problem as firefox is fine and refreshes to changes straight away.

negan07, 2022-05-06:
I didn't encounter any issues with: ie, safari, edge, chrome, mozilla, mozilla ESR

I made the code modification some time ago and since that the recurrent reports of old version cached have stopped...

Linden1, 2022-05-06:
I certainly shouldn't have a cache issue, I very rarely use chrome for network suff. I've always preferred firefox for that.
If you're going to push a new monolithic with the regresses then I think we should wait for that to do anything else with this. My problem is gone so I can't debug anymore really.

negan07, 2022-05-06:
I'm waiting to push the monolithic because I would like to see the evolution of this minor issue, but it's hard if I can't reproduce the situation involved.

@Shockeh: any news about this ?

Linden1, 2022-05-07:
Question - if I delete nvram then install official netgear v84 firmware will all the ancistrus stuff be gone? Or is there another way of cleaning it out other than removing one package at a time? I want to address the other issue of missing menu items.

Linden1, 2022-05-07:
Issue - Ancistrus menu items missing from [post #2](https://github.com/negan07/ancistrus/issues/91#issuecomment-1116687514).
I could only replicate by installing the firmware using the rescue image method (startup.sh)

Replicated by:
1. reset nvram
2. flash latest rescue image
3. add ancistrus using the startup.sh script.

Observation:
1. 'Core Setup' menu populated
2. 'Management' 'Networking' Storage' and 'General' menus are empty.
3. Rootfs partition space used 80%

I can deduce from this some items are left out to keep the rootfs usage down, and it's up to the user to install any packages they need.

negan07, 2022-05-07:
This is not related to browser cache retain

https://github.com/negan07/ancistrus/wiki/Rescue-firmware-&-startup.sh#behaviour

Point number 5: 
`Install the essential packages including [ancistrus-core](https://github.com/negan07/ancistrus/wiki/ancistrus-core-package) and [opkg](https://github.com/negan07/ancistrus/wiki/opkg-package) itself to flash.
`
The essential packages are: `zlib openssl curl ancistrus-core utelnetd opkg
`

Where: `zlib openssl curl utelnetd` are already present on vendor firmware, but older and with some configuration differences so they are replaced.

`ancistrus-core opkg
`

are new

`The `startup.sh` has not been created to install all the packages starting from a vendor firmware.
Each one can decide to install its useful packages, so the menus are void at the beginning because packages are not installed yet.