negan07/ancistrus #92: Feature request: DHCP address reservation extension (dnsmasq)

ComputaBloke, 2022-10-23:
Hi @negan07,

Now my router is hub for lots of WiFi IOT devices, I've hit the limit of 64 DHCP IP Address static reservations.
Do you believe it might be possible to maintain an extended list of DHCP reservations (perhaps extra 65-128)?
I noticed the device is using 'dnsmasq' internally, and so perhaps a DHCP feature could be added similar to Ancistrus's UI for 'Wake On LAN' extension, or via an additional config file?

Example I found on dnsmasq config file, might look like this:
```
# Static IPs
dhcp-host=00:00:5e:00:53:42,thedoctor,192.168.0.10
dhcp-host=00:00:5e:00:53:01,00:00:5e:00:53:02,tardis,192.168.0.22
dhcp-host=00:00:5e:00:53:08,sonicscrewdriver,192.168.0.23
dhcp-host=00:00:5e:00:53:10,satellite5,192.168.0.32
```

Related, I noticed Ancistrus allows us to maintain the standard /etc/hosts file, but exposing more of the 'dnsmasq' back end might even make Ancistrus capable of providing ad-blocking features similar to PiHole!

negan07, 2022-10-23:
Not using dnsmasq internally, the reservation is made through a heavily customed version of udhcpd
The limitation is probably made by `setup.cgi` owner's code
It should be overriden maybe adding manually nvram's array of subsettings `dhcp_reserved` then restarting dhcp service with `rc dhcpd restart`
```
nvram get | grep dhcp_reserved
dhcp_reserved=DEADBEEF10.0.0.3DE:AD:BE:EF:00:00
```
fields are: hostname\2lanip\2macaddr\1
and so on
Unfortunately, there's nothing to do with sub/subfield & cline (`anc nvram` should be implemented specifically for this) so this can be made only editing manually exported nvram file with an hexeditor

the array fields are for each reservation:
\0x01HOSTNAME\0x02IP\0x02MAC\0x01....
first reserv. 0x01 is omitted
last 0x01 is omitted
variable must end with 0x00 (NULL)

reimport nvram then
`nvram commit`
`rc dhcpd restart`
take a look and make a try

negan07, 2022-12-22:
any news about this stuff ?

ComputaBloke, 2024-04-27:
Sorry, no. The D7000 was no longer able to meet requirements for me, and Netgear not released any new updates in a few years, so I decided to get a more powerful Asus AX4200 (which is openwrt-compatible in future).
Thank you for your amazing support for such a long time, Ancistrus is a great project!

closed 2024-04-27 by ComputaBloke