negan07/ancistrus #93: Router crashing daily

wholegraincrouton, 2022-12-14:
Hey, I've installed the latest version available from the releases page, however my router is now crashing daily rather than just whenever I leave the house with my phone connected to the wifi.
The logs seem to be reset whenever the router is restarted, and the router needs to be restarted in order to reach the administration portal to read them.
Is this a common issue? If so, how do I fix it or gather logs which persist across restarts?

m0lz, 2022-12-19:
It may take negan07 a while to notice/respond to this topic ..
.. In the meantime at least I can assure you its not a common problem. The latest version has been running flawlessly since it was released about half a year ago.

negan07, 2022-12-22:
Which kind of connection do you have ?
Does regressing to the original firmware fix the issue ?

closed 2023-02-25 by negan07