negan07/ancistrus #96: Weird issue when switching a client device between wired and wireless interfaces.

Dronec, 2023-04-03:
I have D7000 downstairs and another AP upstairs that are connected via Ethernet. Few other devices are connected to D7000 via Eth as well.
The issue is that when I connect my wireless client device to the AP, it can see and ping wired devices connected to D7000 immediately. When I use the same device and connect wirelessly to D7000, it immediately sees all the wirelessly connected devices, but starts seeing wired ones only after 8-10min.
The issue is apparently in switching the same MAC between wired and wireless interfaces because when I use a client device with randomized MAC, everything's working fine and pings immediately regardless of where I connected
[image]

negan07, 2023-04-04:
Maybe it's not related to D7000 behaviour.

Dronec, 2023-04-20:
> Maybe it's not related to D7000 behaviour.

Tested it with multiple devices, same result. Certainly has something to do with OSI#2: the issue disappeared when I moved the Upstairs router to the different subnet.

closed 2023-05-15 by negan07