negan07/ancistrus #98: Any chance to support Optware/Entware?

Jishun, 2023-08-06:
should have found this awesome project earlier 
thanks!

By going though the issue list I realized that it might be impossible to leverage openwrt package source, though 
optware/entware use uclibc. it'd be good to support optware then enjoy the rick packages

negan07, 2023-08-23:
`uclibc` are only one of the lib compatibilities required.
Many more things must be verified and modified to obtain working packages.
To use them as they are, a deep software reorganization is required, keeping in mind that original kernel core cannot be skipped due to source code lack.

tommaso82, 2025-02-15:
Has anyone had the opportunity to understand if on a hardware level the R7000 is identical to the D7000? (apart from the modem section), In this way we could directly use OpenWRT on our D7000 with some minimal adjustments, maybe we should understand if they have the same bootloader...